doonny/PipeCNN #132: Proper procedure to generate input and output golden reference file

mAxwong1014, 2020-04-10:
Dear prof Wang,
I read through the procedure given by you but still cannot get the idea in generating both files, your detailed  guides are mostly appreciated for me! Thanks in advance.

mAxwong1014, 2020-04-17:
dear prof @donny, I would appreciated it if you can help me, thank you.

doonny, 2020-04-17:
Hi, the mean_data file is obtained from this repo:  https://github.com/pmgysel/alexnet-forwardpath

mAxwong1014, 2020-04-17:
thanks for replying prof @doonny ,
if I want to train my data, how can I generate this mean data? Based on my understanding, the mean_data binaryproto generated in caffe cannot be used directly, I also look through the paper several times but didnt find the answer. Your help is highly appreciated!!! Thanks in advance.

doonny, 2020-04-18:
Please refer the user's manual of Caffe. I remember they have some instruction on setting up data set for training.  Here, we only focus on hardware implementation

mAxwong1014, 2020-04-18:
thanks for replying prof @doonny.
based on my understanding, the hardware implementation is based on ilsvrc12 imagenet competition, I downloaded the imagenet_mean.binaryproto file and try it on pipecnn but the accuracy is poorly performed. May I know any significant change needed to be done on the  imagenet_mean.binaryproto to generate the mean_data,dat? Sorry to bothering you again.

doonny, 2020-07-05:
We have not made any changes to the mean file.