doonny/PipeCNN #133: Compilation de1soc board

sergio14890, 2020-04-14:
Hey guys!
I dont have a de10soc , but i have a de1soc(board).
I would like to know if anyone has tried to implement this fantastic work in it.
Do you think it is possible to implement on the de1soc board?
I downloaded the work, but how should I generate the .aocx files? Since there are quite a few .cl files

sergio14890, 2020-04-15:
When i execute 'makefile' i have this errors:
[](https://i.ibb.co/sgnhbN0/pipe.png)

sergio14890, 2020-04-16:
Hey guys!
I solve this erros, changing some makefile settings.
However, i have one more error!
You know also about it?

`Compiler Command: aoc -v -report -seed=3 -I device/RTL -L device/RTL -l rtl_lib.aoclib -board-package=C:\intelFPGA\17.1\hld\board\de1_soc -I C:\intelFPGA\17.1\hld/include/kernel_headers -g device/conv_pipe.cl -o conv.aocx
c:/Users/sergi/Downloads/PipeCNN-master/project/device/conv_pipe.cl:1094:16: error: attribute takes one argument
__attribute__((max_work_group_size(1,1,LRN_MAX_LOCAL_SIZE))) // (x,y,z)
               ^
1 error generated.
`

doonny, 2020-04-17:
v17.1 is not supported. Try v19.1

sergio14890, 2020-04-17:
Doonny you have a bsp19.1 for de1soc board?
I read some comments here that users use the de1soc board like me.
Or do you have an older version of pipeCNN compatible with bsp17.1 that I can make available?

Alhad1994, 2020-04-24:
Hi sergio,
I have the exact same error! how did you solve it

sergio14890, 2020-04-24:
Hey Alhad1994.
You can give me your contact?

Alhad1994, 2020-04-24:
yes sure its alhadp007@gmail.com
I am working on this project  and really would appreciate your help on some errors and topics

nazarihome, 2020-05-14:
Hi, 
I have the same problem, I was wondering if you figured out the solution.
`~/Desktop/PipeCNN-master/project$ make
aoc -v -report -seed=3  -I device/RTL -L device/RTL -l rtl_lib.aoclib -board-package=/tools/argus/intelFPGA/19.1/hld/board/trasic/de5net -I /tools/argus/intelFPGA/19.1/hld/include/kernel_headers -g device/conv_pipe.cl -o conv.aocx
Resolving library filenames to full paths
  lib_path = device/RTL
    lib_file = rtl_lib.aoclib
Resolved rtl_lib.aoclib to /home/argus/Desktop/PipeCNN-master/project/device/RTL/rtl_lib.aoclib
  lib_path = .
aoc: Environment checks are completed successfully.
aoc: Cached files in /var/tmp/aocl/ may be used to reduce compilation time
You are now compiling the full flow!!
aoc: Selected default target board de5net_a7
aoc: Running OpenCL parser....
/home/argus/Desktop/PipeCNN-master/project/device/conv_pipe.cl:1094:16: error: attribute takes one argument
__attribute__((max_work_group_size(1,1,LRN_MAX_LOCAL_SIZE))) // (x,y,z)
               ^
1 error generated.
Error: OpenCL parser FAILED.
Refer to conv/conv_pipe.log for details.

Makefile:174: recipe for target 'conv.aocx' failed
make: *** [conv.aocx] Error 1
`

Alhad1994, 2020-05-14:
__attribute__((max_work_group_size(1,1,LRN_MAX_LOCAL_SIZE))) // (x,y,z) !! Remove 1,1 and compile again!

ngohieutruong, 2020-10-07:
> **attribute**((max_work_group_size(1,1,LRN_MAX_LOCAL_SIZE))) // (x,y,z) !! Remove 1,1 and compile again!

hello, 
I have a problem, I built SDCard OpenCL 16.1 and copied run.exe, conv.aocx. when run on de1soc, I get an error: error while loading shared libraries: libacl_emulator_kernel_rt.so: cannot open shared object file: No such file or director
Can you help me fix it? thank you!

doonny, 2020-10-09:
you could not run emulator on the FPGA board,  emulator is only support on X86 machine

siowedwin, 2021-03-21:
@sergio14890 do you manage to implement on de1?

sergio14890, 2021-03-21:
Yes

siowedwin, 2021-03-21:
@sergio14890 can i get your contact?