doonny/PipeCNN #135: version `GLIBC_2.17' not found (required by ./run.exe)

sergio14890, 2020-05-01:
Hey guys..
Can someone help me update glibc? Now i have 2.15, but i required 2.17..
https://i.ibb.co/Yc68P4p/1234.png

HemantaIngle, 2020-05-08:
Hi Donny and Sergio i am facing the same issue  can you please help 
[image]

sergio14890, 2020-05-08:
Hey hermantalngle, i dont can solve this problem :(
I am waiting for someone to help me too, if you find out please let me know

HemantaIngle, 2020-05-08:
> Hey hermantalngle, i dont can solve this problem :(
> I am waiting for someone to help me too, if you find out please let me know

Hey Sergio thanks for your reply, please let me know if you get around with the problem , and i will do the same if i find a solution.

sergio14890, 2020-05-09:
> > Hey hermantalngle, i dont can solve this problem :(
> > I am waiting for someone to help me too, if you find out please let me know
> 
> Hey Sergio thanks for your reply, please let me know if you get around with the problem , and i will do the same if i find a solution.

Hey heman, As soon as I get a solution I will contact you, if you find it do the same, thank you very much

Alhad1994, 2020-05-10:
Compiler Error: Unrecognized function call: mult_add_fix8bx4 check this issue! Someone had this error earlier . but has not posted the solution! I am assuming that we have to upgrade the linux version form 16.04 to 18 which will solve the problem. Another fix can be to upgrade the libs which is painful according to many forums! @doonny could you suggest a better way to deal with this error!

HemantaIngle, 2020-05-10:
I just wondered if the machinekit image for a DE0-Nano-SOC board will run on a DE10-Nano?

HemantaIngle, 2020-05-10:
While it's possible to do, there are some disadvantages:

If you make any mistake in the process, you will render the system un-bootable. Better have recovery media ready and know how to use it to manually mount your root filesystem and repair it.
While GLIBC is highly backward compatible, existing programs that depend on undefined behavior may break.
Your best bet is to run this new program in a chroot environment or in a docker container.

Your next best bet is to install GLIBC-2.27 in parallel with system GLIBC, as described in e.g. this answer.

If you do decide to update the system GLIBC to the new version, the least error-prone way to do so is to download a new package, boot the system from recovery media, mount root filesystem on e.g. /mnt, and unpack the new version on top of it.

sergio14890, 2020-05-14:
Hemantalngle ,

> Embora seja possível, existem algumas desvantagens:
> 
> Se você cometer algum erro no processo, você tornará o sistema não inicializável. É melhor ter a mídia de recuperação pronta e saber usá-la para montar manualmente o sistema de arquivos raiz e repará-lo.
> Embora o GLIBC seja altamente compatível com versões anteriores, os programas existentes que dependem de comportamento indefinido podem ser interrompidos.
> Sua melhor aposta é executar esse novo programa em um ambiente chroot ou em um contêiner de docker.
> 
> Sua próxima melhor aposta é instalar o GLIBC-2.27 em paralelo com o sistema GLIBC, conforme descrito, por exemplo, nesta resposta.
> 
> Se você decidir atualizar o GLIBC do sistema para a nova versão, a maneira menos propensa a erros é fazer o download de um novo pacote, inicializar o sistema a partir da mídia de recuperação, montar o sistema de arquivos raiz em, por exemplo, / mnt e descompactar a nova versão no topo disso.

Hey heman, you can solve the problem?

HemantaIngle, 2020-05-14:
@sergio14890 let me take a look.

sergio14890, 2020-05-15:
@Alhad1994 @doonny you can solve the problem?

doonny, 2020-05-16:
You could upgrade to ubuntu 18.04

sergio14890, 2020-05-16:
Update linux in my desktop ?

doonny, 2020-07-05:
Yes, upgrade the GCC to the required version. you could find more information in the SDK user's mannual for v19.3

GUUTA, 2020-08-02:
./run.exe: /lib/libm.so.6: version `GLIBC_2.29' not found (required by ./run.ex)
./run.exe: /lib/libc.so.6: version `GLIBC_2.17' not found (required by ./run.ex)
./run.exe: /lib/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by .)
./run.exe: /lib/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by)
How do I fix this  problem?

closed 2020-12-17 by doonny