doonny/PipeCNN #137: quartus_map ended unexpectedly

Alhad1994, 2020-05-06:
[image]
I am getting this error and I have tried the solution provided by Intel but that does not work! More over I have enough space which I verified. Let me know what solution works??

mustard-seed, 2020-05-06:
Have you looked inside _quartus_sh_compile.log_ for the exact error? It is usually a really verbose document, so just search for "ERROR"

Alhad1994, 2020-05-07:
[image]
Hi , the log error does not provide any additional information. I have tried the Intel's official solution for which says 
Workaround/Fix

To work around this problem if a self-referential assignment, remove the self-referential assignment from the RTL.

To work around the insufficient memory based on Quartus version, replace the installed version of Quartus with the version that matches your system properties.  

I am confused as to what the system version should be!
This problem is fixed in all Pro versions of Quartus.

mustard-seed, 2020-05-07:
Another speculation: when you @Alhad1994 said that "there is enough space", did you also check RAM? It might be possible that your machine doesn't have enough RAM or there is a cap on how much RAM a process can use. Quartus_map might get killed if there is not enough RAM (see this: https://forums.intel.com/s/question/0D50P00003yyRK4SAM/error-293007-current-module-quartusmap-ended-unexpectedly?language=en_US)

doonny, 2020-05-07:
I think you do not have enough RAM. Make sure there are at least 64GB memory for large design.  Or you could set small values for both vec_size and lane_num to reduce the cost on RAM.

Alhad1994, 2020-05-07:
Thank you @doonny  and @mustard-seed . I checked the RAM and I had alloted 4 GB which was not sufficient. I changed vec_size and lane_num to 4 and bumped my RAM to 8 GB on vitual machine which solved the error.

closed 2020-05-08 by Alhad1994