doonny/PipeCNN #138: When i try to make file the following error was generated. How I can fix this problem?

GUUTA, 2020-05-07:
make
../common/src/AOCLUtils/opencl.cpp: In function ‘_cl_program* aocl_utils::createProgramFromBinary(cl_context, const char*, _cl_device_id* const*, unsigned int)’:
../common/src/AOCLUtils/opencl.cpp:388:22: warning: ignoring attributes on template argument ‘cl_int’ {aka ‘int’} [-Wignored-attributes]
  388 |   scoped_array<cl_int> binary_status(num_devices);
      |                      ^
/usr/bin/ld: skipping incompatible /home/guta/intelFPGA/18.1/hld/host/arm32/lib/libalteracl.so when searching for -lalteracl
/usr/bin/ld: skipping incompatible /home/guta/intelFPGA/18.1/hld/board/c5soc/arm32/lib/libintel_soc32_mmd.so when searching for -lintel_soc32_mmd
/usr/bin/ld: cannot find -lintel_soc32_mmd
/usr/bin/ld: skipping incompatible /home/guta/intelFPGA/18.1/hld/host/arm32/lib/libelf.so when searching for -lelf
collect2: error: ld returned 1 exit status
Makefile:77: recipe for target 'bin/host' failed
make: *** [bin/host] Error 1

doonny, 2020-05-08:
for ARM processor, you need cross platform g++ compiler to do the job.

GUUTA, 2020-05-09:
Doonny/Pipecnn
Thank you!

I have fixed an early problem. I'm trying to run your project to see the
effect. But, I can't  fix  the problem like this
aoc -v -report -profile -o device/RTL -L device/RTL -l  rtl_lib.aoco
rtl_lib.aocl  -I /home/guta/intelFPGA/18.1/hld/include/kernel_headers -g
device/pipe.cl -v conv.aocx
aoc: Warning: no argument provided for the option -profile, will enable
profiling for all kernels by default
Resolving library filenames to full paths
  lib_path = device/RTL
    lib_file = rtl_lib.aoco
  lib_path = .
    lib_file = rtl_lib.aoco
Error: Cannot find the following specified library files: rtl_lib.aoco
Makefile:139: recipe for target 'con.aocx' failed
make: *** [conv.aocx] Error 1

thank You for your time and consideration.

warm regards
Guta

doonny, 2020-05-14:
Please carefully read the instructions provided in this repo.
You need to compile the RTL library first.

datnguyen263, 2020-12-21:
@doonny i try to make with board DE1_SoC but error:

aoc: Linking with IP library ...
Compiler Error: Couldn't achieve user-specified II for loop at c:\Users\PipeCNN\project_intel/c:/Users/PipeCNN/project_intel/device/conv_pipe.cl:751
Error: Verilog generator FAILED.
Refer to conv/conv_pipe.log for details.

make: *** [conv.aocx] Error 1

Thank you for your time and consideration.

doonny, 2020-12-29:
v18.1 is not supported, please v20.1

datnguyen263, 2020-12-29:
> v18.1 is not supported, please v20.1

https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=205&No=836&PartNo=4

I dont have v20.1. How about v16.0 or v14.0?

doonny, 2021-01-03:
I am affraid not.