doonny/PipeCNN #139: When I try to change num_compute_units the following error generated. What is cause this? and how I fix it?

GUUTA, 2020-05-26:
Channel Pairing Type Does not match.
Channel bias_ch needs one chan_read_altera and one chan_write_altera.
Multiple chan_read_altera/chan_write_altera with the same channel ID may cause this problem.
system_integrator: custom_ic_impl.cpp:8889: void custom_ic::System::build_channel_system(): Assertion `0' failed.
Aborted (core dumped)
Error: System integrator FAILED.
Refer to conv/My_Kernel.log for details.

doonny, 2020-07-05:
have you solved the problem ?

GUUTA, 2020-07-16:
Dear Doonny

I haven't solved the problem still now?
However I try to take as comments the channel and it work. But I can't
believe it.
Because the result is not good.
If you solved the problem share with me.
Thank you for your consideration and time.
Sincerely,
Guta