doonny/PipeCNN #141: yolo make file and missing files in host directory

kutenai, 2020-06-01:
The Makefile in the ./yolo directory is looking for some host files.

These files:
```
HOST_SRCS = ./host/main_yolo_3.8.cpp ../common/ocl_util.cpp ../common/timer.cpp ./host/utility/region.cpp ./host/utility/box.cpp ./host/utility/showboxes.cpp
```

But they do not exist in the yolo directory. I did find the ./host directory in ./project, but the files listed above are not present their either.

kutenai, 2020-06-02:
I did a little more digging and it looks like you added all of that logic on the latest commit. 
The main project is located in the 'project' directory, so the yolo was and is just a reference at this point. 

It would be great if you could share enough to make it a complete project.

doonny, 2020-07-05:
We will make a update in the following weeks.  Thx for the reminds.