doonny/PipeCNN #144: OPENCV libraries for arm

sergio14890, 2020-07-07:
Hey @doonny  and everyone!
I need download opencv libraries to compile in arm. Can someone tell me a place or tutorial where i can see how to do it?

mingyi136, 2020-07-09:
@sergio14890 you may have a look on OpenCV documentation: 
https://docs.opencv.org/2.4/doc/tutorials/introduction/linux_install/linux_install.html

To summarize up, this is the method that I used to install OpenCV on this Linux SD card image of DE1-SoC from https://software.intel.com/content/www/us/en/develop/topics/fpga-academic/learn/tutorials.html

```
git clone https://github.com/opencv/opencv.git
git checkout tags/2.4.13.3       //as I'm using OpenCV 2.4.13.3 due to CMAKE version available

//Install required packages
[compiler] sudo apt-get install build-essential
[required] sudo apt-get install cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev
[optional] sudo apt-get install python-dev python-numpy libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libjasper-dev libdc1394-22-dev

cd opencv
mkdir release
cd release
cmake -D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_INSTALL_PREFIX=/usr/local ..
make -j2
make install
```

sergio14890, 2020-07-10:
@mingyi136 you know also about this error?
I'm try solved it, but i don't can.
[image]

mingyi136, 2020-07-10:
@sergio14890 which command that you are running? Is that to build opencv for ARM?

sergio14890, 2020-07-10:
Yes my big friend, I'm trying to build opencv for arm.
I run this command:  cmake -DSOFTFP=ON -DCMAKE_TOOLCHAIN_FILE=../opencv/platforms/linux/arm-gnueabi.toolchain.cmake ../opencv

I run this command in the Ubuntu 18.04 to build the opencv for arm.
And then I pass openCV to windows and run it on Intel soc eds command shell

mingyi136, 2020-07-10:
Hi @sergio14890 , I'm not familiar with your method.

But for me, I have ran the commands (to build opencv) that I have mentioned before directly on ARM.

sergio14890, 2020-07-10:
Hmm ok.
One more question, you can connect your fpga to internet? 
If so, could you explain to me how you do it? Is that I can only connect to a home computer-fpga network
Thank you so much!

mingyi136, 2020-07-10:
Yes, you may connect the board to Internet!

For my own way, I have shared my desktop WiFi with DE1-SoC board via LAN Cable 
(https://www.instructables.com/id/How-to-share-Internet-over-Ethernet-Cable/)

Alternatively, you may also try to download opencv folder in dekstop, then transfer it to ARM (to be built later) using scp command (via LAN cable)

sergio14890, 2020-07-10:
Nice, your help was very good! Thank you so much!

sergio14890, 2020-07-10:
Hey @mingyi136 one last time.
I tried to do what you told me.

I have my computer connected to a Wi-Fi network.
And I connected my computer to the de1soc board via 1 ethernet cable.
I think I set everything up well, but when i try search on google in board, i have error.

[image]
[image]

mingyi136, 2020-07-10:
You are most welcome, @sergio14890 

I suggest you to check the connection again by typing 'ifconfig' command in the terminal (ARM).
You should see 'inet addr' value displayed on 'eth0'. If not, you may try reconnect it by typing 'dhclient eth0'

sergio14890, 2020-07-13:
Hey @mingyi136  one more time!
I tried to do what you told me.
But, when i do '**_make_**', in 66% i have this error:
-> fatal error: can\u2019t write PCH file: No space left on device compilation terminated.

I have image de1_soc board on a 16GB micro SD, but when the image is mounted it only uses about 7GB.
I think i need increase the size of the partition /dev/root with about 9Gb that were not used.

How can I do that, you know something?
[image]

sergio14890, 2020-07-13:
[image]

sergio14890, 2020-07-17:
Hey @mingyi136 and @doonny, i need your help ...
I'm having a hard time connecting my fpga to the internet.
I have the following configuration: 

                   **wi-fi**                                  **ethernet cable**
Router     **------->**     My Computer          **------->**              De1-soc Board

In configuration My wi-fi i have enable : properties -> share -> select 2 boxs
[image]

And this is IP config:
[image]

When i connect FPGA via ethernet cable i have this automatic ip:
[image]

But, the network stay: unidentified network and public network.
[image]

In putty i confirate:
ifconfig eth0 169.254.27.152 netmask 255.255.0

But when i try connect internet, i dont have a sucess.
I need help, thank you for sugestions

mingyi136, 2020-07-17:
Have you tried 'dhclient eth0' in ARM terminal? 
Does 'inet addr' value display on 'eth0' section after you enter 'ifconfig' command in ARM terminal?

[image]

sergio14890, 2020-07-17:
[image]

When i do "dhclient eth0", the terminal is stuck.
I have this wi-fi adaptor in my home: [https://www.conrad.com/p/logilink-wl0084e-wi-fi-dongle-usb-20-150-mbps-987133](url).
I'm try connect to my internet with command:   ./connect_wpa my_internet_name password, but doesn't work

mingyi136, 2020-07-17:
Perhaps you can try this, it works for me:

1. First, make sure your Wi-Fi internet connection is unshared

[image]

2. Highlight both Ethernet (the one to board) and Wi-Fi, right-click and click 'Bridge-Connections' 

[image]

3. Wait for a while, then right-click the Ethernet and select 'Remove from Bridge'

[image]

4. Finally, make the Wi-Fi internet connection shared

[image]

5. Try again with 'dhclient eth0' and 'ifconfig'

sergio14890, 2020-07-17:
I did exactly what you asked for, and I got this result ..
[image]

sergio14890, 2020-07-17:
[image]

Do you know if it is normal to have only 1 led access on the ethernet cable port?

sergio14890, 2020-07-23:
Thanks @mingyi136 finally, i solved all problems.
I have this kit TRDB-D5M, you know if it's possible open this web-cam in Arm?

mingyi136, 2020-07-23:
Glad to hear that @sergio14890 

I didn't try TRDB-D5M before, perhaps you can have a try on that

sergio14890, 2020-07-24:
And with a USB camera have you ever tried? @mingyi136