doonny/PipeCNN #152: how do I solve wrong result ?

GUUTA, 2020-10-08:
For VGG16 and AlexNet
Start verifying results ...                                                     
Item=0 is wrong (result=0.000000, golden_ref=-6.000000)                         
Item=1 is wrong (result=0.000000, golden_ref=3.000000)                          
Item=2 is wrong (result=0.000000, golden_ref=-8.000000)                         
Item=3 is wrong (result=0.000000, golden_ref=-9.000000)                         
Item=4 is wrong (result=0.000000, golden_ref=-5.000000)                         
Item=5 is wrong (result=0.000000, golden_ref=-1.000000)                         
Item=6 is wrong (result=0.000000, golden_ref=-12.000000)                        
Item=7 is wrong (result=0.000000, golden_ref=7.000000)                          
Item=8 is wrong (result=0.000000, golden_ref=10.000000)                         
Totally 941 Wrong Results

doonny, 2020-10-09:
it seems that your program has not correctly load the image.dat file, so everything was zeros

GUUTA, 2020-10-09:
> it seems that your program has not correctly load the image.dat file, so everything was zeros

So how I would solve it? I was put all information in the /data folder. But it doesn't access it.