doonny/PipeCNN #153: A error when compiling the project

davidtong-300, 2020-10-16:
Hey guys, I got an error when tried to compile the program. It looks like a problem that the project cannot find the lib file, but I tried many ways still not working. 
[image]
My Ubuntu version is 16.04
Quartus and Open CL SDK for FPGA  is 18.1
OpenCV is 2.x.
Does anyone can give me any suggestions？ Thank you for your help.

datnguyen263, 2020-11-09:
Do you solved the problem?

doonny, 2020-11-20:
It seems you are building for ARM processor, make sure you have correct cross-compiling env set up .

datnguyen263, 2020-12-10:
> It seems you are building for ARM processor, make sure you have correct cross-compiling env set up .

How to check cross-compliling env setup? Tks for your answer!