doonny/PipeCNN #154: Failed to get device name size

davidtong-300, 2020-10-24:
@thinkoco Hi thinkoco, we are trying to run PipeCNN on the de1soc board. We used your all-in-one linux image and de1soc_vga BSP. For host computer we used sdk18.1. We successfully compiled it on the host computer but when we tried to execute it on the board this error showed up. Do you have any suggestions for this? BTW, for your image providing, we used the init_opencl.17.1.sh, does that matter? Thanks a lot.
[image]

thinkoco, 2020-10-27:
@davidtong-300 
try the new  [sdcard image](https://drive.google.com/file/d/1YbNVyZNJ654OGFZCXptz1Vg0f7xC2NlZ/view?usp=sharing) , and source the  init_opencl.18.1.sh (sdk 18.1.0, not 18.1.1)

davidtong-300, 2020-10-28:
@thinkoco 
Hi, thinkoco, I used your new sdcard image and sourced the init_opencl.18.1.sh, but the problem was still the same.
I check the [aocl diagnose], and it said the device is de1soc_sharedonly_vga. It was different from the BSP in my host computer, which is de1soc_shareonly. 
I tried to replace the de1soc_sharedonly folder in board/terasic/de1soc/hardware to the one from your website de1soc_sharedonly_vga, but still not working. 
I also checked the device name by aocl diagnose and aocl binedit conv.aocx list, make sure that they are both de1soc_sharedonly_vga.
I am confused now, don't know what to do next. 
Do you have any suggestions? Thanks a lot for your kind help.

thinkoco, 2020-10-28:
@davidtong-300 
have you program the fpga by rbf or aocx  firstly? 
1.you can update the opencl.rbf (in sdcard) with top.rbf (in build project). 
2. [run aocl program](https://github.com/thinkoco/c5soc_opencl/blob/master/documents/HowToReconfigureFPGA.md#fpga-reconfiguration) in serial port terminal
When building the hardware, you may add `--board de1soc_sharedonly_vga ` in you makefile to specify the board name.
Also, you can build and run the demo apps to check the runtime environment

closed 2020-12-17 by doonny

datnguyen263, 2021-02-28:
[image]
Dear @doonny 
The same problem and i cant fix it. Can u help me? tks for answer