doonny/PipeCNN #158: how to config addition_precision_config in layer_config.h

WenyeLiu, 2020-11-30:
Hi,

I am going to deploy a resnet like model on pipecnn. However, the accuracy is extremently low comparing with the results in the simulation framework.
I underestand that the precision_config is used to set the quantiazation bitwidth for weights and input/output feature maps.
May I ask what is the usage of the "addition_precision_config" in layer_config.h. It seems used to convert between char and float data type for batchnorm layer and eltwise layer. Is there any explanation for those parameters in "addition_precision_config"

Many thanks.

doonny, 2021-03-23:
will update the instructions later

WenyeLiu, 2021-04-21:
Thanks doonny. Looking forward your tips.