doonny/PipeCNN #161: How can i run this project in ZCU102 with SD card?

zhaobr340104, 2021-04-09:
I have been puzzled about some steps with ZCU102 implementation. Looking forward to reply for questions below. Thank you!
1.
[image]
Which file should I attached to the "EDGE_COMMON_SW" parameter? I cannot find the 2020.1ZCU102 directory in your project. 

2.
I also attempted to create a Petalinux project after exporting the xsa file from Vivado2020.1. And I set the  "EDGE_COMMON_SW" parameter to my petalinux project. However, when I run "make image", the compile shows below ERROR.
[image]

SO, my question is about how to make a complete SD card image for ZCU102 board in this project.
Thank you very much for your answer! 
My Wechat account is zbr330232872. Welcome to contact with me for all of candidates in this project!

doonny, 2021-04-20:
you need to download the common imagefiles(rootfs) from Xilinx's websit or generate it by using petalinux

zhaobr340104, 2021-04-24:
Thank you very much for your help! Now I can 'make host' and 'make image' successfully!
I can see these files generated by 'make image'.
 
[image]

[image]
 
But when I put all these files in SD card,  the petalinux system cannot start.
When I checked the Xilinx official information, I found that there was a missing file called 'image.ub' for SD card.
 Do I have to generate image.ub or get this file in somewhere? 
Looking forward to your advice! Thank you!

zhaobr340104, 2021-04-25:
I can run this project on ZCU102 with SD card now.
Step1: copy all of files in the "sd_card" directory into SD card fat32 partition.
Step2 : copy the Xilinx official rootfs system into SD card ext4 partition.
Thank you for your advice and your projcet!

closed 2021-04-25 by zhaobr340104