doonny/PipeCNN #165: Running PipeCNN on U200

Alsumbjip, 2021-09-20:
Hi,
I had tried to run PipeCNN on Xilinx Alveo U200, I'd modified Makefile and config file for U200 and TEST_CASE_1, but it's not worked for HW-Emulation (SW-Emulation is working).

Modified PipeCNN had stopped at CoreConv, and I guess it was waiting for stream input about 'weight_in' and 'data_in'.

And I found arguments 'output_num' and 'conv_loop_cnt'  value was zero in waveform profile data.

memRead, memWrite module was working correctly, and arguments 'frac_din', 'frac_w' in coreConv had correct value, but only arguments 'output_num'~'frac_dout' was wrong.

I'd like to get a hint for fixing this problem.

I modified:
 - project_xilinx/Makefile: CNN_TYPE->TEST_CASE_1, DEVICE->xilinx_u200_xdma_201830_2
 - config_sp.u50: comment/uncomment stream connect for TEST_CASE_1

I working at:
- Ubuntu 18.04
- Vivado 2020.1 / Vitis 2020.1 /  Vitis HLS 2020.1 / XRT version 2.9.317

Thank you.

doonny, 2021-10-30:
Could you run sw emulation ?

lloo099, 2021-11-24:
> 

@Alsumbjip Hi, do u deploy PipeCNN successfully in U200? I am also want to try it. thanks,

lloo099, 2021-11-24:
Hi Prof. Dong, @doonny  I also try to deploy your NN on U200. sw_emu is well implement on U200 platform. But hw and hw_emu are failed, you may check below information:
(I just modify platform to U200 in makefile, and doesn't change config which means it use U50 config for my U200. As I don't know how to define ur config file for U200, and could you help me with that? Many thanks!)
```
INFO: [v++ 60-1306] Additional information associated with this v++ link can be found at:
	Reports: /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/reports/link
	Log files: /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/logs/link
INFO: [v++ 60-1657] Initializing dispatch client.
Running Dispatch Server on port:45307
INFO: [v++ 60-1548] Creating build summary session with primary output /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/conv.xclbin.link_summary, at Wed Nov 24 22:11:00 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Wed Nov 24 22:11:00 2021
Running Rule Check Server on port:46179
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/reports/link/v++_link_conv_guidance.html', at Wed Nov 24 22:11:01 2021
INFO: [v++ 60-895]   Target platform: /tools/Xilinx/Vitis/2020.1/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm
INFO: [v++ 60-1578]   This platform contains Device Support Archive '/tools/Xilinx/Vitis/2020.1/platforms/xilinx_u200_xdma_201830_2/hw/xilinx_u200_xdma_201830_2.dsa'
INFO: [v++ 60-1302] Platform 'xilinx_u200_xdma_201830_2.xpfm' has been explicitly enabled for this release.
INFO: [v++ 60-629] Linking for hardware emulation target
INFO: [v++ 60-423]   Target device: xilinx_u200_xdma_201830_2
WARNING: [v++ 60-889] User-specified kernel frequency for ID 0 is the same as the default frequency 300 MHz, so it will be ignored
INFO: [v++ 60-1332] Run 'run_link' status: Not started
INFO: [v++ 60-1443] [22:11:08] Run run_link: Step system_link: Started
INFO: [v++ 60-1453] Command Line: system_link --xo /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/memRead.xo --xo /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/coreConv.xo --xo /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/memWrite.xo --xo /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/maxPool.xo -keep --config /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/int/syslinkConfig.ini --xpfm /tools/Xilinx/Vitis/2020.1/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm --target emu --output_dir /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/int --temp_dir /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link
INFO: [v++ 60-1454] Run Directory: /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/run_link
INFO: [SYSTEM_LINK 60-1316] Initiating connection to rulecheck server, at Wed Nov 24 22:11:09 2021
INFO: [SYSTEM_LINK 82-70] Extracting xo v3 file /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/memRead.xo
INFO: [SYSTEM_LINK 82-70] Extracting xo v3 file /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/coreConv.xo
INFO: [SYSTEM_LINK 82-70] Extracting xo v3 file /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/memWrite.xo
INFO: [SYSTEM_LINK 82-70] Extracting xo v3 file /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/tmp.hw_emu/maxPool.xo
INFO: [SYSTEM_LINK 82-53] Creating IP database /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/_sysl/.cdb/xd_ip_db.xml
INFO: [SYSTEM_LINK 82-38] [22:11:09] build_xd_ip_db started: /tools/Xilinx/Vitis/2020.1/bin/build_xd_ip_db -ip_search 0  -sds-pf /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/emu/xilinx_u200_xdma_201830_2_emu.hpfm -clkid 0 -ip /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/iprepo/xilinx_com_hls_memRead_1_0,memRead -ip /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/iprepo/xilinx_com_hls_maxPool_1_0,maxPool -ip /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/iprepo/xilinx_com_hls_memWrite_1_0,memWrite -ip /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/iprepo/xilinx_com_hls_coreConv_1_0,coreConv -o /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/_sysl/.cdb/xd_ip_db.xml
INFO: [SYSTEM_LINK 82-37] [22:11:11] build_xd_ip_db finished successfully
Time (s): cpu = 00:00:03 ; elapsed = 00:00:02 . Memory (MB): peak = 1439.266 ; gain = 0.000 ; free physical = 9123 ; free virtual = 19872
INFO: [SYSTEM_LINK 82-51] Create system connectivity graph
INFO: [SYSTEM_LINK 82-102] Applying explicit connections to the system connectivity graph: /home/enai/Desktop/project/enai0/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/cfgraph/cfgen_cfgraph.xml
INFO: [SYSTEM_LINK 82-38] [22:11:11] cfgen started: /tools/Xilinx/Vitis/2020.1/bin/cfgen  -sc memRead_1.bias_out:coreConv_1.bias_in -sc memRead_1.weight_out:coreConv_1.weight_in -sc memRead_1.data_out:coreConv_1.data_in -sc coreConv_1.conv_out:batchNorm_1.conv_in -sc coreConv_1.bypass_out:memWrite_1.bypass_in -sc batchNorm_1.bn_out:memWrite_1.bn_in -dmclkid 0 -r /home/enai/Desktop/project/enai0/PipeCNN/project_xi [... cut ...]
ERROR: [SYSTEM_LINK 82-79] Unable to create system connectivity graph
INFO: [v++ 60-1442] [22:11:13] Run run_link: Step system_link: Failed
Time (s): cpu = 00:00:06 ; elapsed = 00:00:05 . Memory (MB): peak = 1346.410 ; gain = 0.000 ; free physical = 9149 ; free virtual = 19897
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.

```

> Could you run sw emulation ?

lloo099, 2021-11-29:
Update on U200: just compiled VGG-16 on U200 successfully. But haven't tried ResNet. Pls tell me if anyone run ResNet on U200. Thanks,

zyt1024, 2023-03-21:
> Hello,If I run on the u200, my config_sp.u200 file was copied from config_sp.u50. What needs to be modified? I don't know how to  modify the config_sp.u200 file, and could you help me with that? Many thanks!

zyt1024, 2023-03-21:
> Update on U200: just compiled VGG-16 on U200 successfully. But haven't tried ResNet. Pls tell me if anyone run ResNet on U200. Thanks,

Hello,If I run on the u200, my config_sp.u200 file was copied from config_sp.u50. What needs to be modified? I don't know how to modify the config_sp.u200 file, and could you help me with that? Many thanks!

zyt1024, 2023-03-29:
> Update on U200: just compiled VGG-16 on U200 successfully. But haven't tried ResNet. Pls tell me if anyone run ResNet on U200. Thanks,

Hello, I also runnedVGG-16 on U200 successfully. but the performance on the U200 is not very good. How about your performance? this is my.
[image]