doonny/PipeCNN #168: make fpga error for xilinx

justinjiaxinghu, 2021-11-21:
`ERROR: [CFGEN 83-2291] --sc tag applied with invalid slave kernel instance: batchNorm_1
ERROR: [CFGEN 83-2291] --sc tag applied with invalid master kernel instance: batchNorm_1
ERROR: [CFGEN 83-2298] Exiting due to previous error
ERROR: [SYSTEM_LINK 82-36] [23:37:29] cfgen failed
Time (s): cpu = 00:00:04 ; elapsed = 00:00:04 . Memory (MB): peak = 1693.863 ; gain = 0.000 ; free physical = 26333 ; free virtual = 111169
ERROR: [SYSTEM_LINK 82-62] Error generating design file for /nethome/jhu357/Documents/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/cfgraph/cfgen_cfgraph.xml, command: /tools/reconfig/xilinx/Vitis/2020.2/bin/cfgen  -sc memRead_1.bias_out:coreConv_1.bias_in -sc memRead_1.weight_out:coreConv_1.weight_in -sc memRead_1.data_out:coreConv_1.data_in -sc coreConv_1.conv_out:batchNorm_1.conv_in -sc coreConv_1.bypass_out:memWrite_1.bypass_in -sc batchNorm_1.bn_out:memWrite_1.bn_in -dmclkid 0 -r /nethome/jhu357/Documents/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/_sysl/.cdb/xd_ip_db.xml -o /nethome/jhu357/Documents/PipeCNN/project_xilinx/build.hw_emu/link/sys_link/cfgraph/cfgen_cfgraph.xml
ERROR: [SYSTEM_LINK 82-96] Error applying explicit connections to the system connectivity graph
ERROR: [SYSTEM_LINK 82-79] Unable to create system connectivity graph
INFO: [v++ 60-1442] [23:37:29] Run run_link: Step system_link: Failed
Time (s): cpu = 00:00:16 ; elapsed = 00:00:16 . Memory (MB): peak = 1585.230 ; gain = 0.000 ; free physical = 26371 ; free virtual = 111207
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:212: recipe for target 'build.hw_emu/conv.xclbin' failed
make: *** [build.hw_emu/conv.xclbin] Error 1`

OS:  Ubuntu 18.04.6 LTS (Bionic Beaver)

lloo099, 2021-11-24:
what platform you use?

justinjiaxinghu, 2021-11-24:
> 

x86 

below is a snippet from makefile

```
#set source and target for device
#select the host archecture  x86|aarch32|aarch64
PLATFORM = x86
#PLATFORM = aarch64
#PLATFORM = aarch32

#choose the BSP (with the config files) you would like to use
DEVICE := xilinx_u50_gen3x16_xdma_201920_3
CONFIG_SP := config_sp.u50
```

lloo099, 2021-11-25:
> > what platform you use?
> 
> x86
> 
> below is lscpu
> 
> ```
> Architecture:        x86_64
> CPU op-mode(s):      32-bit, 64-bit
> Byte Order:          Little Endian
> CPU(s):              40
> On-line CPU(s) list: 0-39
> Thread(s) per core:  2
> Core(s) per socket:  10
> Socket(s):           2
> NUMA node(s):        2
> Vendor ID:           GenuineIntel
> CPU family:          6
> Model:               79
> Model name:          Intel(R) Xeon(R) CPU E5-2630 v4 @ 2.20GHz
> Stepping:            1
> CPU MHz:             1200.393
> CPU max MHz:         3100.0000
> CPU min MHz:         1200.0000
> 
> BogoMIPS:            4399.73
> Virtualization:      VT-x
> L1d cache:           32K
> L1i cache:           32K
> L2 cache:            256K
> L3 cache:            25600K
> ```
> 
> below is a snippet from makefile
> 
> ```
> #set source and target for device
> #select the host archecture  x86|aarch32|aarch64
> PLATFORM = x86
> #PLATFORM = aarch64
> #PLATFORM = aarch32
> 
> #choose the BSP (with the config files) you would like to use
> DEVICE := xilinx_u50_gen3x16_xdma_201920_3
> CONFIG_SP := config_sp.u50
> ```

seems no big issue, and u may run sw_emu first.

justinjiaxinghu, 2021-11-25:
what do you mean no big issue? Are you saying that I should make with sw_emu first and then do hw_emu?

lloo099, 2021-11-25:
> what do you mean no big issue? Are you saying that I should make with sw_emu first and then do hw_emu? 

Yes. I run sw_eni successfully but failed in he

justinjiaxinghu, 2021-11-25:
> > what do you mean no big issue? Are you saying that I should make with sw_emu first and then do hw_emu?
> 
> Yes. I run sw_eni successfully but failed in he

I'm confused. are you saying you're having the same issue?

lloo099, 2021-11-26:
> > > what do you mean no big issue? Are you saying that I should make with sw_emu first and then do hw_emu?
> 
> > 
> 
> > Yes. I run sw_eni successfully but failed in he
> 
> 
> 
> I'm confused. are you saying you're having the same issue? I am also failing on make hw_emu after running sw_emu

- 
Yes, u May check #165

justinjiaxinghu, 2021-11-27:
update on this issue:

I actually think it has something to do with the ifdef statements. I switched to VGG and for some reason had to manually comment out the stream variables corresponding to  the resnet model for things to compile properly. additionally, the config file needs to be modified when you switch networks as well. I hope this can be helpful to others, but I also hope these issues will be worked out  in a future  commit

lloo099, 2021-11-27:
> update on this issue:
>
>
>
> I actually think it has something to do with the ifdef statements. I switched to VGG and for some reason had to manually comment out the stream variables corresponding to [...]

Thanks much. Could you provide modified codes to help us?

doonny, 2021-11-27:
For xilinx platform, only VGG is supported now.

justinjiaxinghu, 2021-11-29:
> Thanks much. Could you provide modified codes to help us?

Yes I will add them all here soon.

lloo099, 2021-12-03:
> Yes I will add them all here soon.

```
./run.exe conv1.xclbin
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************

Platform: Xilinx
Totally 1 device(s) are found
  Using Device 0: xilinx_u200_xdma_201830_2
Device OpenCL Version: OpenCL 1.0
Device Max Compute Units: 0
Device Max WorkGroup Size: -1
Device Max WorkItem Size: -1
Device Global Memory Size: 65536 MBytes
Device Local Memory Size: 16 KBytes
Device Max Clock Freq: 500 Mhz

138455872 total weights read 
1024 total output reference read 

150528 bytes image data read from binary files

Executing Layer 1:

Launching single work-item kernel winbuffer

Launching single work-item kernel Conv

Launching single work-item kernel MemWr

```
Hi, I encounter a problem about Memwr. When I run this .exe, the system is no response at Kernel MemWr for long time. Do u know how to solve it? It seems no feedback from device

justinjiaxinghu, 2021-12-03:
> ```
> ./run.exe conv1.xclbin
> ***************************************************
> PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs
> ***************************************************
>
> Platform: [...]

I didn't encounter this error so I'm unsure. I also used the u50 board. It seems you have jumped over all the other compilation hurdles, though.

Anyways, I'll outline the changes I had to make to make VGG compile and run (at least for hardware emulation).

If you are using the default makefile provided, you shouldn't need to make any changes other than the following for the makefile (also just change the flag from sw_emu to your desired setting as specified in the user instructions). 
```
HOST_SRCS = ./host/main_aoc.cpp ../../common/ocl_util.cpp ../../common/timer.cpp
#use Xilinx OpenCL C++ host binding
#HOST_SRCS = ./host/main.cpp ../../common/xcl2.cpp ../../common/timer.cpp
HOST_OBJS = $(HOST_SRCS:%.cpp=%.o)
HOST_INCS = ../../common
```
change to

```
HOST_SRCS = ./host/main_aoc.cpp ../common/ocl_util.cpp ../common/timer.cpp
#use Xilinx OpenCL C++ host binding
#HOST_SRCS = ./host/main.cpp ../../common/xcl2.cpp ../../common/timer.cpp
HOST_OBJS = $(HOST_SRCS:%.cpp=%.o)
HOST_INCS = ../common
```

The config file:
Make sure to uncomment out the lines under VGG connection and comment out the ones for Resnet

The code:
Under project_xilinx/devices

In memWrite.cpp
```
#ifdef RESNET
				uchar  bypass,
				uchar  pool_pad,	  //RESNET need pad,set to 1,other CNN set 0
#endif
...
...
...
#ifdef RESNET
				hls::stream<k2k_data_xlane>     &bn_in,
				hls::stream<k2k_data_xlane>     &bypass_in
#else
				hls::stream<k2k_data_xlane>     &conv_in
				//hls::stream<k2k_sync>           &pool_sync_out
#endif
```
Change to
```
/*#ifdef RESNET
				uchar  bypass,
				uchar  pool_pad,	  //RESNET need pad,set to 1,other CNN set 0
#endif*/
...
...
...
/*#ifdef RESNET
				hls::stream<k2k_data_xlane>     &bn_in,
				hls::stream<k2k_data_xlane>     &bypass_in
#else/*
				hls::stream<k2k_data_xlane>     &conv_in
				//hls::stream<k2k_sync>           &pool_sync_out
//#endif
```

I am unsure why these have to specifically be commented out. If I didn't, I got some invalid port errors. 

Under project_xilinx/host

In main_aoc.cpp

The user defined library paths are incorrect
```
// user defined library
#include "ocl_util.h"
#include "timer.h"
```
change to

```
// user defined library
#include "../../ocl_util.h"
#include "../../timer.h"
```

hope this helps some people, hopefully i didnt miss anything

zyt1024, 2023-03-21:
> ```
> ./run.exe conv1.xclbin
> ***************************************************
> PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs
> ***************************************************
>
> Platform: [...]

How did you solve this problem? I also encountered this problem