doonny/PipeCNN #173: Software emulation OpenCL runtime error, Xilinx/Vitis

sysuresh, 2022-04-22:
We are getting the following error when running software emulation ``` make emu ``` with Xilinx Vivado and Vitis - what could be the cause?
```
***************************************************
PipeCNN: An OpenCL-Based FPGA Accelerator for CNNs 
***************************************************
ERROR: Unable to find the desired OpenCL platform.
```

doonny, 2022-05-20:
Please check the Vitis installation and enviroment settings.

zyt1024, 2023-03-29:
> Please check the Vitis installation and enviroment settings.

Hello, I would like to ask, if I am using the U200 platform, does this place in the makefile need to be modified?
``` shell
#set the Linux related tools
#EDGE_COMMON_SW := /opt/petalinux/2020.1
EDGE_COMMON_SW := /home/fpga/Xilinx/petalinux/2020.1ZCU102
#EDGE_COMMON_SW := /home/fpga/Xilinx/petalinux/2020.1ZC706```