doonny/PipeCNN #176: Make host error, no opencl header files, ZCU102/Vitis/Xilinx

qiuzeyu, 2023-03-03:
Hello, I am trying to compile this project for the Xilinx ZCU102 board on Ubuntu18.04.2 64-bit. According to the **_README.md_**  under **_documents_** folder, the **_make fpga_** command is successfully executed, and the **_conv.xclbin_** file is generated. But when I execute the **_make host_** command, the following error occured:
[image]

 I have installed Vitis 2020.1 and Xilinx Runtime(XRT)2020.1 on Ubuntu, but there is no **_opencl.h_** file in the XRT installation path：
[image]

I also installed the opecnCL library for Ubuntu and found **_opencl.h_** in the `/usr/include/CL` path:
[image]

Then I executed the following command and compile again:
**`export CPLUS_INCLUDE_PATH=/opt/xilinx/xrt/include/:/usr/include`**
new errors appeared:
[image]

I don't understand what's wrong. How can I get the right **_opencl.h_** file please？ And is there a more specific tutorial on compiling this project for Xilinx ZCU102 board? 

This is my **Makefile**, please help me.
[Makefile.txt](https://github.com/doonny/PipeCNN/files/10880514/Makefile.txt)

Thank you.

zyt1024, 2023-03-21:
hello,你的解决问题了嘛?

zyt1024, 2023-03-21:
我的也报错..

zyt1024, 2023-03-21:
你的解决了嘛，我的也报错，不知道为什么
[image]

qiuzeyu, 2023-03-22:
> 你的解决了嘛，我的也报错，不知道为什么 [image]

这个是因为common文件夹下的ocl_util.cpp没有编译，把Makefile中编译host的命令改成Intel的再make host：
[image]

qiuzeyu, 2023-03-22:
此错误已解决，问题是缺少petalinux SDK造成的。部署PipeCNN到ZCU102上，除了安装Vitis和Petalinux工具之外，还需要从Xilinx官网下载安装ZynqMP通用镜像，镜像压缩包内包含了制作SD卡系统的Linux内核及根文件系统。解压后按照压缩包内的readme说明在指定路径下（需修改Makefile中EDGE_COMMON_SW的路径与此相同）执行sdk.sh安装SDK，SDK中提供了编译host程序所需的OpenCL头文件：
[image]

closed 2023-03-24 by qiuzeyu